roboticslibrary/rl #90: Black Screen on Ubuntu22.04 and unable to see normally

jack416, 2024-05-16:
use in VM ubuntu22.04 gcc/g++ version 11.4.0  cmake version 3.22.1

I have installed RL via homebrew and when I run the first example

and the other examples too I get only a black windows 

but save as png have rendered content , like this

run：
/usr/local/bin/rlPlanDemo /usr/local/share/rl-0.7.0/examples/rlplan/unimation-puma560_boxes_rrtConCon.xml

running results：
[image]

How can I do to get things working ?

MarceloEO, 2024-05-17:
Hi Jack,
Try to change the render buffer type to single:

right button on the rendering scene -> draw styles -> Render buffer type -> Single 

I hope this will help you!

jack416, 2024-05-20:
> Hi Jack, Try to change the render buffer type to single:
> 
> right button on the rendering scene -> draw styles -> Render buffer type -> Single
> 
> I hope this will help you!

Thank you for your response. I have implemented the above functionality using thefunction.
Thanks！

closed 2024-05-20 by jack416