roboticslibrary/rl #91: How to implement MoveL and MoveC?

xianjimli, 2024-05-19:
I would like the end-effector of the robotic arm to move in straight lines and arcs instead of the shortest path. Could you give me an implementation idea? Thank you.

rickertm, 2024-05-19:
For implementing linear and circular trajectories in operational space, you need to define trajectories for the position and orientation of the end effector in operational space. These can be combined into a 6D position+orientation vector and transformed to the robot's configuration space via the inverse of the Jacobian matrix.

The classes rl::math::Polynomial and rl::math::Spline include a few functions for interpolation position vectors, they also include a specialization for quaternions that can be used for the orientations. rl::math::Circular3 can be used to interpolate a circular segment through three points.

xianjimli, 2024-05-20:
Thank you so much

closed 2024-05-20 by xianjimli