roboticslibrary/rl #94: rlPrmDemo problem?

northeasternLiu, 2024-09-18:
dear author,I want to study robotPlan,but rlPrmDemo has the problem,how to solve this problem?
[image]

rickertm, 2024-09-19:
Could you please provide more information on what compiler and which compiler version you are using?

I assume you are using the 0.7.0 binaries? Does the error also appear with the latest third-party dependencies (see [rl-3rdparty](https://github.com/roboticslibrary/rl-3rdparty) repository and related binaries under [Releases](https://github.com/roboticslibrary/rl-3rdparty/releases)) and the latest version of RL on the master branch?

northeasternLiu, 2024-09-20:
I use  visualstudio2017, using 0.7 binaries
[image]

rickertm, 2024-10-10:
If the translation is correct, then the file `..\..\src\rl\plan\Debug\rlplansd.lib` cannot be opened and there are 2 failed steps in total, not just this one? Can you please provide the error message for the other step, including a translation to English?

Furthermore, does this issue only show when building the Debug version or also for the Release version? Did you test it with a current version of the third-party dependencies and the latest RL version from the master branch as mentioned above?