roboticslibrary/rl #97: problem with rlCoachKin and rlPlanDemo

saiwing, 2025-02-14:
I installed rl following https://www.roboticslibrary.org/tutorials/install-ubuntu/ successfully, but when I run the examples, I got empty display with rlCoachKin and rlPlanDemo, while other examples had no problem

/usr/bin/rlCoachKin /usr/share/rl-0.7.0/examples/rlsg/unimation-puma560_boxes.xml /usr/share/rl-0.7.0/examples/rlkin/unimation-puma560.xml
[image]

/usr/bin/rlPlanDemo /usr/share/rl-0.7.0/examples/rlplan/unimation-puma560_boxes_rrtConCon.xml
[image]

rickertm, 2025-02-16:
Thank you very much for the report and the analysis. We've had a few similar reports (#90) that were fixed by setting the render buffer type to single.

Could you please provide some more information regarding the graphics card, graphics driver, and display server (Xorg or Wayland) you're using, maybe together with relevant output from `glxinfo`?

saiwing, 2025-02-28:
I used ubuntu in VM virtual machine，(VMWorksation 16 pro 16.2.3)
```
$ glxinfo
name of display: :0
display: :0  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    ...
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
client glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    ...
GLX version: 1.4
GLX extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    ...
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: VMware, Inc. (0x15ad)
    Device: SVGA3D; build: RELEASE;  LLVM; (0x405)
    Version: 23.2.1
    Accelerated: no
    Video memory: 1MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.1
    Max compat profile version: 4.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.0
OpenGL vendor string: VMware, Inc.
OpenGL renderer string: SVGA3D; build: RELEASE;  LLVM;
OpenGL core profile version string: 4.1 (Core Profile) Mesa 23.2.1-1ubuntu3.1~22.04.3
OpenGL core profile shading language version string: 4.10
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
    GL_AMD_conservative_depth, GL_AMD_draw_buffers_blend, 
    ...

OpenGL version string: 4.1 (Compatibility Profile) Mesa 23.2.1-1ubuntu3.1~22.04.3
OpenGL shading language version string: 4.10
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL extensions:
    GL_AMD_conservative_depth, GL_AMD_draw_buffers_blend, 
   ...
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 23.2.1-1ubuntu3.1~22.04.3
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
OpenGL ES profile extensions:
```
```
$ lspci | grep -i vga
00:0f.0 VGA compatible controller: VMware SVGA II Adapter
```
```
$ echo $XDG_SESSION_TYPE
x11
```