roboticslibrary/rl #99: How to correctly find and link RL on Windows (DLLs + CMake integration)

gsisinna, 2025-06-04:
Hi, and thanks for your work on the Robotics Library!

I'm trying to use RL in a custom C++ project on Windows and would like to ask a few questions about the correct setup process:

**Finding the RL package:**
Is there an official or recommended way to find the RL installation on Windows via CMake (find_package() or similar)? Should I rely on environment variables, CMake config files, or manually specify paths?

**Linking the DLLs:**
After building or downloading RL, what is the best way to link against the generated DLLs in a Windows environment? Are there import libraries (.lib files) provided alongside the DLLs that I should use when linking?

**Precompiled binaries:**
I noticed that there may be binaries available from [the official site](https://www.roboticslibrary.org/). Are these intended for use in custom projects, and if so, is there any guidance or documentation on integrating them?

Any guidance or best practices for setting up RL in a Windows-based project would be greatly appreciated!

Thanks again!

gsisinna, 2025-06-09:
Thanks again, @rickertm! I've opened an initial "new port" issue [here](https://github.com/microsoft/vcpkg/issues/45830) in the official [microsoft/vcpkg](https://github.com/microsoft/vcpkg/) repository.

Next, I'll begin exploring how to integrate this library with **vcpkg** to enhance its usability and streamline dependency management.

gsisinna, 2025-06-10:
I'm happy to report that I successfully built RL on Windows using vcpkg for dependency management. With some minor adjustments, the process was fairly smooth, and using vcpkg really helped standardize the setup across different machines. Most of RL's dependencies are already available in the vcpkg registry. However, I did encounter a few exceptions:

- ATIDAQ
- cifX
- libdc1394

This reinforces the idea that better integration with package managers like vcpkg can significantly reduce the friction of building RL, especially on Windows where dependency resolution can be challenging. Adding RL itself to vcpkg would be a natural next step, and I'd be happy to contribute or collaborate on that effort.

Let me know how I can help with any of the above—especially vcpkg integration or testing future tags on Windows.

I leave here the `vcpkg.json` and `vcpkg_configuration.json` that I used to download and install the necessary dependencies:

**vcpkg.json**
```json
{
  "name": "rl",
  "version-string": "0.7.0",
  "description": "A self-contained C++ library for rigid body kinematics and dynamics, motion planning, and control",
  "dependencies": [
    "boost",
    "eigen3",
    "libxml2",
    "libxslt",
    "nlopt",
    "bullet3",
    "coin",
    "ccd",
    "fcl",
    "ode",
    "pqp",
    "solid3",
    "soqt",
    "simage",
    "vcpkg-tool-ninja"
  ]
}
```

You can also override the versions for each individual library and freeze them by using the current tag.

**vcpkg_configuration.json**
```json
{
  "default-registry": {
    "kind": "git",
    "baseline": "984f9232b2fe0eb94f5e9f161d6c632c581fff0c",
    "repository": "https://github.com/microsoft/vcpkg"
  },
  "registries": [
    {
      "kind": "artifact",
      "location": "https://github.com/microsoft/vcpkg-ce-catalog/archive/refs/heads/main.zip",
      "name": "microsoft"
    }
  ]
}
```

You can easily automate vcpkg integration in your CMake to test this:
```cmake
if(WIN32)
  if(DEFINED ENV{VCPKG_ROOT})
    set(CMAKE_POLICY_DEFAULT_CMP0077 NEW)
    set(X_VCPKG_APPLOCAL_DEPS_INSTALL ON)
    message(STATUS "Using Vcpkg from: $ENV{VCPKG_ROOT}")
    set(CMAKE_TOOLCHAIN_FILE
        "$ENV{VCPKG_ROOT}/scripts/buildsystems/vcpkg.cmake"
        CACHE STRING "")
  else()
    message(WARNING "Vcpkg not found. Set VCPKG_ROOT environment variable.")
  endif()
endif()
```

rickertm, 2025-06-15:
Thank you for testing RL against the current vcpkg dependencies. The missing dependencies are only used for selected hardware support (force/torque sensors, fieldbus, firewire camera) and likely not required for most users.

We'll look into creating a new tag, that should make packaging a bit easier.